sparkfun/SparkFun_Bio_Sensor_Hub_Library #6: Arduino Nano Crash after First LED flash

lawsonfulton, 2020-05-21:
Hi there,

I followed [this](https://learn.sparkfun.com/tutorials/sparkfun-pulse-oximeter-and-heart-rate-monitor-hookup-guide) guide with the exception that I'm running on an Arduino Nano.

Everything seems to work well when I run the example getting the following output:
```
Sensor started!
Configuring Sensor....
Sensor configured.
Loading up the buffer with data....
Heartrate: 0
Confidence: 0
Oxygen: 0
Status: 0
```

However, as soon as I put my finger on the device, the LED flashes once, and the Arduino stops responding. Nothing gets printed to the screen. I then need to unplug the arduino before it starts working again. 

I'm an experience programmer, but new to working with low level electronics and I'm not sure how to proceed with debugging this. If anyone has an idea of what I could try next, it would be much appreciated!

alexmstahl, 2020-05-21:
I had the same problem when I first tried this sensor with a Teensy. It turned out to be a power supply issue. The LED draws enough current that the 3.3V output from my Teensy was dropping to 2.7v and causing a reset.

I fixed it by using an external 3.3v regulator instead of using the 3.3v output from the board.

I hope this helps,

-alex (Another user)

edspark, 2020-06-03:
I'll close this out for now. @lawsonfulton please reopen if you had another question, though perhaps our [forums](https://forum.sparkfun.com/) are best for these types of questions.

lawsonAGMT, 2020-06-03:
Thanks for the info. I haven't had a chance to try it out yet but I suspect that is the issue.

lawsonAGMT, 2020-06-08:
For those interested, switching over to using an external voltage regulator fixed my problem.

Although, oddly, it only worked when I was powering the board and the regulator from the same 12V source. It would not work when I powered the board over USB, and the regulator from 12V...