sparkfun/SparkFun_Bio_Sensor_Hub_Library #9: ARDUINO LIBRARY FOR MAX30101 VERSION D FOR MEASURING BLOOD PRESSURE

parminder4499, 2020-06-21:
I am using sparkfun's spo2 sensor board(MAX30101 version A) for measuring spo2 and heart rate. I want to measure blood pressure as well from MAX30101 version D which has firmware by MAXIM integrated for measuring BP. I hereby wanted to ask you if you can modify the Arduino library for MAX30101 version A, so that it can also measures Blood pressure when using version D of the sensor?
If that library is build up then we can freely use MAX30101's different versions for making products using sparkfun's library.
Thanks

santaimpersonator, 2020-06-25:
Duplicate issue. See: https://github.com/sparkfun/SparkFun_Bio_Sensor_Hub_Library/issues/10

Also closing.

edspark, 2020-12-15:
After looking through the datasheet, this could be done in an afternoon. I'll try to add it soon.

edspark, 2021-01-06:
Added in latest [pull request](https://github.com/sparkfun/SparkFun_Bio_Sensor_Hub_Library/pull/17), I may have to do some additional tweaking but the functions are there.

closed 2021-01-06 by edspark