sparkfun/SparkFun_Bio_Sensor_Hub_Library #10: ARDUINO LIBRARY FOR MAX30101 VERSION D FOR MEASURING BLOOD PRESSURE

parminder4499, 2020-06-25:
I am using sparkfun's spo2 sensor board(MAX30101 version A) for measuring spo2 and heart rate. I want to measure blood pressure as well from MAX30101 version D which has firmware by MAXIM integrated for measuring BP. I hereby wanted to ask you if you can modify the Arduino library for MAX30101 version A, so that it can also measures Blood pressure when using version D of the sensor? 
If that library is build up then we can freely use MAX30101's different versions for making products using sparkfun's library.
Thanks

santaimpersonator, 2020-06-25:
@parminder4499 I am moved this issue to the [MAX32664 version of the Arduino library](https://github.com/sparkfun/SparkFun_Bio_Sensor_Hub_Library) and will be closing the issue for the reasons below.

Unfortunately, as previously mentioned, we don't utilize that version of the chip and therefore we probably aren't going to provide any development in the foreseeable future. However, if you make are able to make the necessary modifications, feel free to submit a pull request so that other users may benefit from your work.

closed 2020-06-25 by santaimpersonator

parminder4499, 2020-06-26:
@santaimpersonator  sure, thanks for the information