sparkfun/SparkFun_Bio_Sensor_Hub_Library #12: Add support for MAX30102 Sensor

ChinmayPJE, 2020-09-10:
Add support for MAX30102 Sensor

ChrisIdema, 2020-12-09:
I have the same issue. It reads raw led values just fine. It detects the finger. But heart rate isn't correct at all.

edspark, 2020-12-15:
Sorry for the late reply, let me see what that would entail.

ChrisIdema, 2020-12-16:
I've created a pull request. I've fixed some bugs to get it to work on the ESP32 and I fixed a bug in the delay of the readFillArray so that function is no longer too slow.

I've got it to work with the MAX30101 and the MAX30102 without making any further modifications! So my issues weren't related to the MAX30102 after all.

The problem I experienced was that the MAX32664 fills its FIFO very quickly. You have to continuously check the FIFO or else it will fill up and first the values will lag behind and then you will get invalid values because there are FIFO errors. I haven't found a way to avoid this. I've created an example that reads it quickly: see my pull request.

edspark, 2021-01-06:
Thanks @ChrisIdema I'll make comments on your pull request as it's a little out of the scope of this issue. @ChinmayPJE this library does not directly support the MAX30101 or the MAX30102, that's the job of the MAX32664 on the product. The MAX32664 has firmware which supports both the MAX30101 and MAX30102.  You can see that in the [datasheet](https://datasheets.maximintegrated.com/en/ds/MAX32664.pdf) under the second paragraph on the first page which reference "Version A", the version on this product.

closed 2021-01-06 by edspark