sparkfun/SparkFun_Bio_Sensor_Hub_Library #18: Using Green LED??

lalitheranti, 2021-01-19:
I’d like to use the green LED (also counts) since that is more effective for use on wrists to detect heart beat. How do I use this library or modify it?

LEIS23, 2022-11-10:
Hi, did you solve this problem?

edspark, 2023-06-15:
Unfortunately this product is designed for fingertip readings only and consequently it does not utilize the green LED. This is because the Bio-Sensor Hub (MAX32664) is configured with "Firmware Version A" which is configured for IR and RED LED input.  However this product uses the MAX30101 which we DO have as a [separate product](https://www.sparkfun.com/products/164740) that has its own [Arduino Library](https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library) which allows you to address the [green LED specifically](https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/blob/72d5308df500ae1a64cc9d63e950c68c96dc78d5/src/MAX30105.cpp#LL551C25-L551C25). 

Good luck!

closed 2023-06-15 by edspark

jayffnorton, 2023-11-29:
The product link seems to be broken, would it work with [this](https://www.sparkfun.com/products/16474) sparkfun product instead which uses a MAX30101 sensor? In the readme it says the library should work with this, but just wanted to double check.

edspark, 2023-11-29:
That's the one!