sparkfun/SparkFun_Bio_Sensor_Hub_Library #19: Sensor stops responding after variable amount of time

kylejohnson, 2021-02-06:
Heyo!  Not sure if this is the right place to post this issue, but I'll start here anyway.

I'm running the sensor on an ESP8266.
After some amount of time (e.g. I've noticed it after 30 minutes, and after 90 minutes), the sensor stops reporting: the red LED stops lighting up, and I can no longer get data back from the sensor.  The only way that I've found to recover is to powercycle the sensor.

I'll post more data the next time that this happens, but wanted to post now while it was fresh on my mind, to see if anyone else has encountered a similar issue.

kylejohnson, 2021-02-06:
I was able to recreate the issue quickly tonight.  While wearing the sensor, it was actively reporting, and then suddenly the LED turned off, and I stopped getting data.

Note that I am using the code from https://github.com/sparkfun/SparkFun_Bio_Sensor_Hub_Library/pull/16, though this same issue happens without those modifications.

Here's a snippet of the serial output from just before to when this happened.  It is when all of the values start returning `0` that the sensor is 'dead'.
```
Heartrate: 78
Confidence: 99
Oxygen: 98
Status: 3

Sending Bio data to InfluxDB.
bpm,confidence=99,status=3 heartrate=78i,oxygen=98i
Sending MPU data to InfluxDB.
mpu ax=-5916i,ay=-7400i,az=-10344i,gx=-15454i,gy=6829i,gz=13308i

Heartrate: 78
Confidence: 99
Oxygen: 98
Status: 3

Sending Bio data to InfluxDB.
bpm,confidence=99,status=3 heartrate=78i,oxygen=98i
Sending MPU data to InfluxDB.
mpu ax=-10508i,ay=-9616i,az=-6244i,gx=1072i,gy=-5422i,gz=-6402i

Heartrate: 0
Confidence: 0
Oxygen: 98
Status: 3

Sending MPU data to InfluxDB.
mpu ax=-6044i,ay=-11440i,az=-11252i,gx=-926i,gy=-107i,gz=1247i

Heartrate: 0
Confidence: 0
Oxygen: 0
Status: 0

Sending MPU data to InfluxDB.
mpu ax=0i,ay=0i,az=0i,gx=0i,gy=0i,gz=0i

Heartrate: 0
Confidence: 0
Oxygen: 0
Status: 0

Sending MPU data to InfluxDB.
mpu ax=0i,ay=0i,az=0i,gx=0i,gy=0i,gz=0i
```

And here is my code.  There's also a few other things going on with this device - I have an LCD which I write the data to, an MPU6050 which I grab data from, and then I upload data to influxdb via UDP.
If I'm doing something wonky, and you're wondering why I'm doing it that way, the answer is: "because I am a c++ noob".

```c++
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <SparkFun_Bio_Sensor_Hub_Library.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_RESET     -1 // Reset pin # (or -1 if sharing Arduino reset pin)

const char ssid[] = "skynet_";
const char pass[] = "fubar";
const byte resPin = 14;
const byte mfioPin = 12;
[... 199 lines cut ...]
    Serial.printf("Status: %d\n", bsts);
    Serial.println();

    if (bsts == 3 && boxygen > 10 && bheartrate > 10) {
      send_to_influxdb(bheartrate, bconfidence, boxygen, bsts);
      write_led(boxygen, bheartrate, bsts, "Yes");
    } else {
      write_led(boxygen, bheartrate, bsts, "No");
    }

    do_mpu();
    
    samples = bioHub.numSamplesOutFifo();
    Serial.println();
  } 

  //clear fifo before delay
  while(samples){
     body = bioHub.readBpm();
     samples--;
  }

  delay(10); 
}
```

Tomisdebom, 2021-05-19:
I'm running into a similar issue and I'm wondering if it's the same as yours. For my application, I need to be able to run the software after the finger is already placed on the sensor. Basically, I place my finger on the sensor, press reset on my Arduino and look at the plot. Whenever I do, the sensor stops responding after around 10 seconds (sometimes shorter). It essentially crashes and is unresponsive after.

[image]

This is the code I'm running:

`
#include <SparkFun_Bio_Sensor_Hub_Library.h>
#include <Wire.h>

// Reset pin, MFIO pin
int resPin = 4;
int mfioPin = 5;

// Possible widths: 69, 118, 215, 411us
int width = 69; 
// Possible samples: 50, 100, 200, 400, 800, 1000, 1600, 3200 samples/second
// Not every sample amount is possible with every width; check out our hookup
// guide for more information.
int samples = 1600; 
int pulseWidthVal;
int sampleVal;

// Takes address, reset pin, and MFIO pin.
SparkFun_Bio_Sensor_Hub bioHub(resPin, mfioPin); 

bioData body; 

// following data: 
// body.irLed      - Infrared LED counts. 
// body.redLed     - Red LED counts. 
// body.heartrate  - Heartrate
// body.confidence - Confidence in the heartrate value
// body.oxygen     - Blood oxygen level
// body.status     - Has a finger been sensed?

void setup(){

  Serial.begin(115200);

  Wire.begin();
  int result = bioHub.begin();

  int error = bioHub.configSensorBpm(MODE_ONE); // Configure Sensor and BPM mode , MODE_TWO also available
  
  error = bioHub.setPulseWidth(width);
  error = bioHub.setSampleRate(samples);

  delay(1000);

}

void loop(){

    // Information from the readSensor function will be saved to our "body"
    // variable.  
    body = bioHub.readSensorBpm();
    Serial.print(body.irLed); 
    Serial.print("  ");
    Serial.println(body.redLed); 
    // Slow it down or your heart rate will go up trying to keep up
    // with the flow of numbers
}`