sparkfun/SparkFun_Bio_Sensor_Hub_Library #26: ESP32 - Gap in raw data

christian-nils, 2023-06-02:
Hi,

Thanks for this library. I bought the oximeter sensor from Sparkfun (https://www.sparkfun.com/products/15219). I am trying to read the IR led count data but there are some periodic gaps in the data. Here is a short excerpt of the Serial monitor: https://pastebin.com/85jETjyG

Here is a graph showing periodic plateaus in the data (which are typically just gaps in data as the ESP32 is just polling the same value until a new value is available).

[image]

I have the same behaviour when reading the data only on interrupt and when the data ready flag is set to 1 (I am using ESP-IDF for that test):

[image]

The code I used when using the Arduino library is the following:

```c
#include <SparkFun_Bio_Sensor_Hub_Library.h>
#include <Wire.h>

// No other Address options.
#define DEF_ADDR 0x55

// Reset pin, MFIO pin
const int resPin = 26;
const int mfioPin = 25;

//#define SDA 21
//#define SCL 22

//#define SDA 27
//#define SCL 14

// Possible widths: 69, 118, 215, 411us
int width = 69; 
// Possible samples: 50, 100, 200, 400, 800, 1000, 1600, 3200 samples/second
// Not every sample amount is possible with every width; check out our hookup
// guide for more information.
int samples = 1600; 
int pulseWidthVal;
int sampleVal;

// Takes address, reset pin, and MFIO pin.
SparkFun_Bio_Sensor_Hub bioHub(resPin, mfioPin); 

bioData body; 

// following data: 
// body.irLed      - Infrared LED counts. 
// body.redLed     - Red LED counts. 
// body.heartrate  - Heartrate
// body.confidence - Confidence in the heartrate value
// body.oxygen     - Blood oxygen level
// body.status     - Has a finger been sensed?

void setup(){

  Serial.begin(115200);

  Wire.begin();
  int result = bioHub.begin();

  int error = bioHub.configSensor(); // Configure Sensor
  
  error = bioHub.setPulseWidth(width);
  error = bioHub.setSampleRate(samples);

  delay(1000);

}

void loop(){

    // Information from the readSensor function will be saved to our "body"
    // variable.  
    body = bioHub.readSensor();
    Serial.println(body.irLed); 
//    Serial.print("  ");
//    Serial.println(body.redLed); 
    // Slow it down or your heart rate will go up trying to keep up
    // with the flow of numbers
}
```

Any idea why I get those periodic gap in the raw data?

Thanks in advance!

edspark, 2023-06-05:
Thanks for the issue, it's on my list and I'll get to this issue if not today, then tomorrow!

christian-nils, 2023-06-05:
Thank you for looking into that. 
From my long investigation during  the last days, I can see the following points (in the case you use MFIO as interrupt line):

- It seems that the data are transmitted every 0.125s for some reason.
- When the configured sampling frequency is increased the more data point you find in each chunk.
- If you use 1600Hz as sampling rate, you need to make sure to set the I2C bus frequency to 400kHz
- To retrieve raw data, you need to make sure the Wrist algo and the AGC algo are disabled
- When using the wrist algo and AGC algo, the sampling rate of the raw data is always 100Hz (25Hz when no finger detected)
- The structure of the data is interesting, the 10-12 bytes from the payload seem to correspond to a sample counter (I cannot find any reference of such counter in any datasheet as those bytes should be the LED4 counts). This is true for both configurations; sensor only or sensor + algo (one_mode, did not check the other configurations):

[image]
- It seems that by default there is some averaging done on the data points as I retrieve always half of the number of points for the configured sampling frequency (I can see from the MAX30101 datasheet that this is the default anyway, I will try to decrease that number and see if it changes anything).

Anyway, maybe some stuff is wrong (or everything), but I thought that it was worth mentioning those results if it helps. I will use the "counter" from the payload to sync the data. I hope it helps.

edspark, 2023-06-12:
Thanks for the follow up, this may help quite a bit. This library has a few open issues that I'd like to tackle simultaneously as they have a common thread with the ESP32 platform. Unfortunately I got bludgeoned with illness last week and I'll need to do some catch up on my current projects before I get to this. I expect I'll have some time midweek. Talk to you soon.

christian-nils, 2023-06-12:
Thanks for following up :-) I hope you are feeling fine again, take care and good luck with all the catching-up! Let me know if I can help in any ways!

edspark, 2023-06-15:
I suspect it has something to do with the "CMD_DELAY" being significantly reduced in later firmware releases but I still need to test with a board that has the latest firmware which I do not have. This is still ongoing, thank you for your patience.